littleyoda/littleyoda-DCC-Decoder #2: Über die Arduino IDE kann ich nicht kompilieren. Es kommt ein Fehler

murdok80, 2018-07-17:


littleyoda, 2018-07-17:
schau mal auf 
https://github.com/littleyoda/littleyoda-DCC-Decoder/wiki/Entwicklungsumgebung#eclipse-f%C3%BCr-den-esp8266-konfiguieren

Dort sind die 3 Libaries aufgeführt, die du dem Projekt hinzufügen musst.

murdok80, 2018-07-17:
Hab's hinbekommen. Jetzt wär noch der SMD Schaltplan interessant. Die SMD
Platine ist doch für den Anschluss an einen Decoder gedacht.

littleyoda, 2018-07-23:
Es gibt zwei Platinen für die DCC Generierung, die einen Decoder steuern kann.

Eine Platine, die nur ein SMD-Bauteil benötigt:
https://github.com/littleyoda/littleyoda-DCC-Decoder/wiki/USECASE_DCC_Generator_DRV8870

Und eine Platine, die mehr SMD-Bauteile benötigt:
http://gartenbahntechnik.de/download/file.php?id=2783&mode=view

Vergleich der Größen:
http://gartenbahntechnik.de/download/file.php?id=2755&mode=view

Schreib mir einfach eine E-Mail, dann schicke ich dir den Schaltpläne zu.

closed 2018-07-23 by littleyoda

murdok80, 2018-07-23:
Hallo
Ich würde beide mal ausprobieren.  SMD löten ist für mich kein Problem.
Und die 2 Versionen sind ja zum Anschluss an einen bestehenden DCC Decoder.
Wenn ich eine ältere analoge Lok wie z.B. die Stainz umrüsten will brauch
ich ja eine andere Schaltung. Hast du auch eine kleine Platine mit Plan im
dieser Richtung. In der Stainz geht es schon ziemlich eng zu.
Grüße Andre