littleyoda/littleyoda-DCC-Decoder #3: Keine DCC Signale vom Gleis

jakespeed1311, 2018-12-21:
Hallo,
ich habe eine neue Umgebung alles letzter stand Adruino IDE 1.8.8 letzte Lib's.
Mit einem Arduino Uno habe ich meine DCC Eingangsstufe.. Optokoppler siehe Weichendecoder geprüft und das funktioniert. Mein  config file ist angelehnt an den Weichendecoder nur das ich einen Motor steuere, dieser ist aus der Z21 app ( Z21 simulaiton) auch ansprechbar. Toll. Und natürlich aus dem Webinterface auch.

Mein Problem ist das ich keine DCC Signale vom Gleis bekomme um den Decoder zu steuern, im DCCSniffer sehe ich keine Pakete, im Log aber nur die Z21app Befehle.

Nun habe ich gefunden das die NMRA 2.0 lib just bestimmte External Aufrufe nicht mehr nutzt, ok!
Ob das mein Problem ist kann ich nicht sagen.

Der Eingangspin war schon auf D7 bzw D2 beide keine Änderung, am esp kanns auch nicht liegen. Und vom Optokoppler kommen nur max 3,3V, so gehts auch mit dem Uno beim Testen. Einen zweiten ESP8266 hatte ich auch schon geprüft.
Also einzig und alleine der DCC Eingang geht nicht und die Addresse ist auch die Richtige die ich ansteuere. Wobei ich die hoffnung habe der DCC sniffer zeit alles an was am DCC Bus passiert!?
Hat jemand einen Tip ?
Gruß und Danke
Jake

oldnat, 2018-12-21:
Hi Jake!

Sorry, ich bin überfragt... ich nutze eine fertige uns viel einfachere Variante von Sven...

LG Nat

>sent via cellphone<

jakespeed1311, 2018-12-21:
Hallo Nat,
ich glaube das ist mir entgangen.. wo find ich das ? 
Lg Jake

oldnat, 2018-12-21:
Du findest alles on GitHub. And die beste Möglichkeit zum Austausch ist Sven's Forum:

https://forum.open4me.de

LG Nat

>sent via cellphone<

jakespeed1311, 2018-12-21:
Hi,
danke sehr! Hatte ich noch nicht gefunden.
lg
Jake

littleyoda, 2018-12-26:
Komm mal zu uns ins Forum, dürfte einfacher sein.

Ich liefere eine modifizierterte NMRA-Datei mit. Du brauchst keine externe.

littleyoda, 2019-05-10:
Es hat sich heute gezeigt, dass in der NMRA-Lib ein Fehler ist. In meiner Version, die ich mit ausliefere, ist der Fehler korrigiert.

closed 2019-05-10 by littleyoda