littleyoda/littleyoda-DCC-Decoder #13: Link zu Weichendecoder ist fehlerhaft

Donderda, 2024-06-30:
Moin,
der Link https://spurg.open4me.de/wordpress/784/Weichendecoder wirft leider einen 500er. Hast Du die Daten / Informationen noch irgendwo anders? Ich würde mir gerne einen Decoder nachbauen, nur so "komplett aus den Fingern" bekomme ich mir das nicht gesogen :).

LG

littleyoda, 2024-06-30:
Sagst du mir, wo es verlinkt ist? Dann könnte ich den Link anpassen.
https://www.open4me.de/index.php/spurg/weichendecoder/

Bitte daran denken, dass das Config-File nicht mehr stimmt. 
Müsste an den aktuellen Stand angepasst werden.

Duesselklaus, 2024-06-30:
Hallo Sven,

 

Deine – automatisierte? – Mail überrascht und erfreut mich: Bezüglich der Überraschung: Ich kann mich nicht erinnern, in den letzten Monaten da ein Issue gemeldet zu haben, kann es mir aber ansehen. Vielleicht habe ich mal versucht einem Link zu folgen, den es nicht gab – und das System hat es gemerkt. Da gucke ich mal.

 

Die Freude: Ich habe in den vergangenen Wochen sehr oft an Dich gedacht und hätte Dir auch bald mal einen Gruß geschickt. Das ESP32-Drehzahlregler-Projekt ist leider längere Zeit in Wartestellung gegangen – wird aber vermutlich schon in diesem Herbst wieder aufleben. Auch werde ich wohl meine „Aufbauanleitung mir Direktsteuerung des TAMS Decoders“ erweitern können, da ich mal wieder eine Lok umgebaut habe und es aufgefallen ist, dass da ein Hinweis fehlt.

 

Deinen Namen habe ich kürzlich auch in anderem Zusammenhang gelesen: NMRA-DCC für den Arduino. Da habe ich zwei kleine Arduino-Projekte umgesetzt, die auf der Lib basiert, an der du mitgearbeitet hast. Danke dafür!

 

Wenn Du neugierig bist, schreibe ich Dir gern ausführlicher.

 

Grüße aus Düsseldorf

Klaus

 

https://www.open4me.de/index.php/spurg/weichendecoder/

Donderda, 2024-06-30:
Hier in der Readme.md

Donderda, 2024-06-30:
Kannst du mir einen Hinweis geben, wo ich im Code gucken muss, welche Configs möglich sind?

littleyoda, 2024-06-30:
ActionTurnout ist die relevante Klasse zur Ansteuerung von Weichenmotoren.